Ebiroll/qemu_esp32 #6: Use with docker and GitLab CI to run IDF unit tests

lukecyca, 2018-03-24:
This is a really cool project. My first thought was that it could be used with GitLab CI (or another CI) to automatically run [unittests](http://esp-idf.readthedocs.io/en/latest/api-guides/unit-tests.html).

So for example you could have CI automatically running the unittests for your firmware (or library) repo on each checkin, each PR branch, etc.

All the pieces are here, but getting them all strung together is a daunting task. Anybody interested in working on this together?

DavidAntliff, 2018-03-25:
I think that's a good idea. I've been running unit tests with my ESP32 code with a "platform" layer that I can swap out for a POSIX layer to emulate (or at least stub out) certain ESP-IDF functions, but the idea of using QEMU and the ESP-IDF directly appeals greatly.

While I don't have the time to contribute directly, I will be watching with great interest.

Ebiroll, 2018-03-29:
I have previously added the possibility to exit qemu by writing to io adress 0x5F1 on success and to 0x5F0 on failure.
https://github.com/Ebiroll/qemu-xtensa-esp32/commit/92538bd31672bb7b7dc4853b5bd6d1bf93e3b6be

I should put these in a module usable by the unit tests framework so that you don't have to do,
int *quemu_exit=(int *) 0x3ff005f0;
*quemu_exit=0x01;
Whenever there is an assert in the unit tests.

I have also experimented with a posix layer and FreeRTOS, https://github.com/Ebiroll/linux_esp32
Here is a another interesting repository where he has made a complete update to v9 of FreeRTOS
https://github.com/megakilo/FreeRTOS-Sim
Just to let you know.