Ebiroll/qemu_esp32 #8: Error when dumping ROM

jordiUoI, 2018-07-02:
Hi all, I am trying to test the emulator because it is an interesting project for me.
Unfortunately I run into an error at step 2 of the Quick Start, when I try to dump the ROMs.
I have cloned the espressif/esp-idf repository.

My system is Ubuntu 14.04 on a VirtualBox.
Any help will be highly appreciated.
My output when executing `sudo ~/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 -b 921600 -p /dev/tty0 dump_mem 0x40000000 0x000C2000 rom.bin` follows.

> 
> esptool.py  v2.3.1
> Connecting...
> Traceback (most recent call last):
>   File "/home/username/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2637, in <module>
>     _main()
>   File "/home/username/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2630, in _main
>     main()
>   File "/home/username/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 2356, in main
>     esp.connect(args.before)
>   File "/home/username/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 407, in connect
>     last_error = self._connect_attempt(mode=mode, esp32r0_delay=False)
>   File "/home/username/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 365, in _connect_attempt
>     self._port.setDTR(False)  # IO0=HIGH
>   File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 535, in setDTR
>     fcntl.ioctl(self.fd, TIOCMBIC, TIOCM_DTR_str)
> IOError: [Errno 22] Invalid argument

v1993, 2018-08-27:
Are you sure that tty0 is correct port? It is usually a system terminal.

Ebiroll, 2018-08-27:
I also think /dev/tty0 is not correct. You should use the same port as when you program the device.
Another option is to use the (Dumping the ROM with the main/main.c program) section of the README.
If you do this you have to enable   
xTaskCreate(&dump_task, "dump_task", 2048, NULL, 5, NULL); in app_main()

closed 2019-01-03 by Ebiroll