Ebiroll/qemu_esp32 #10: Clarify readme please

greenaddress, 2019-01-15:
I see in the readme one step is to do 'mkdir qemu_esp32' 
but then later the user is asked to clone a repo named qemu_esp32 and it seems both would be in $HOME/esp and would conflict.

I think I am misunderstanding something regarding the above.

Especially so because I don't see logs via telnet and the io.txt log file doesn't seem to show any of my code (gets stuck on wifi read 0x1c) 

I tried both esp-idf master and v3.1.2

thanks!

Ebiroll, 2019-01-15:
You missed the previous step.
cd esp   
git clone git://github.com/Ebiroll/qemu_esp32

When you are finished there will be 2 dirs named qemu_esp
$HOME/esp/qemu_esp
$HOME/qemu_esp
I will clarify this in the document .
$HOME/qemu_esp will contain the binary (qemu) to run.

greenaddress, 2019-01-16:
@Ebiroll seems to work fine now! thanks again. One question: is it possible to enable 4M PSRAM (or more) in qemu?

Ebiroll, 2019-01-16:
Hello, I am not sure if the 4M PSRAM  works.  Will test the examples/01_max_malloc example and see what happens when you enable this. I guess it is in Component config → ESP32-specific → SPI RAM config

Ebiroll, 2019-01-16:
No. qemu did not start. I will see if I can fix this.

E (12) cpu_start: Failed to init external RAM!
abort() was called at PC 0x40080fc5 on core 0

You can set,
  [*]   Ignore PSRAM when not found  
Then it starts but uses 32K RAM
Free heap size 270496 instead of
Free heap size 303768

E (12) spiram: SPI RAM enabled but initialization failed. Bailing out.
I (12) cpu_start: Failed to init external RAM; continuing without it.
I (12) cpu_start: Pro cpu up.
I (12) cpu_start: Single core mode
I (13) heap_init: Initializing. RAM available for dynamic allocation:
I (13) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (13) heap_init: At 3FFB2810 len 0002D7F0 (181 KiB): DRAM
I (13) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (13) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (13) heap_init: At 4008C340 len 00013CC0 (79 KiB): IRAM
I (13) cpu_start: Pro cpu start user code
esp32_i2c_interruptSet: new IRQ val 0x3569540
TBD(pc = 400888db): D:/msys2/home/Olof/esp/qemu-xtensa-esp32/target-xtensa/translate.c:1409
I (269) cpu_start: Starting scheduler on PRO CPU.
I (269) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
NV_INIT_CALLED