Ebiroll/qemu_esp32 #11: MacOS compilation supported?

triple-it, 2019-02-06:
Hi before I dive deep into this very cool/fun project!
Is mac platform supported? Because I tried and immediately receive some errors like:

Error 1:
======
/Users/roderickgroesbeek/esp/qemu/qemu-xtensa-esp32/hw/xtensa/MemoryMapped.cpp:124:45: error: use of
      undeclared identifier 'O_LARGEFILE'
  _file = ::open(filename.c_str(), O_RDWR | O_LARGEFILE);

Error 2:
======
/Users/roderickgroesbeek/esp/qemu/qemu-xtensa-esp32/hw/xtensa/MemoryMapped.cpp:284:19: error: no member named
      'mmap64' in the global namespace; did you mean 'mmap'?
  _mappedView = ::mmap64(NULL, mappedBytes, PROT_READ | PROT_WRITE, MAP_SHARED, _file, offset);
                ~~^~~~~~
                  mmap

Error 3:
======
/Users/roderickgroesbeek/esp/qemu/qemu-xtensa-esp32/hw/xtensa/esp32_sha.c:331:13: error: void function
      'esp_sha_write' should not return a value [-Wreturn-type]
            return 0;
            ^      ~
/Users/roderickgroesbeek/esp/qemu/qemu-xtensa-esp32/hw/xtensa/esp32_sha.c:334:10: error: void function
      'esp_sha_write' should not return a value [-Wreturn-type]
         return 0;
         ^      ~

Fix for Error1 - 2:
============
unkphone-4:qemu-xtensa-esp32 roderickgroesbeek$ git diff
diff --git a/Makefile b/Makefile
index 50b4b3a..8f8dadd 100644
--- a/Makefile
+++ b/Makefile
@@ -1,4 +1,5 @@
 # Makefile for QEMU.
+CFLAGS="$(CFLAGS) -Wno-return-type"
 
 # Always point to the root of the build tree (needs GNU make).
 BUILD_DIR=$(CURDIR)
diff --git a/hw/xtensa/MemoryMapped.cpp b/hw/xtensa/MemoryMapped.cpp
index b3fc4fc..5cf6e1d 100644
--- a/hw/xtensa/MemoryMapped.cpp
+++ b/hw/xtensa/MemoryMapped.cpp
@@ -121,7 +121,8 @@ bool MemoryMapped::open(const std::string& filename, size_t mappedBytes, CacheHi
   // Linux
 
   // open file , O_RDONLY
-  _file = ::open(filename.c_str(), O_RDWR | O_LARGEFILE);
+  //[Unk] _file = ::open(filename.c_str(), O_RDWR | O_LARGEFILE);
+  _file = ::open(filename.c_str(), O_RDWR);
   if (_file == -1)
   {
     _file = 0;
@@ -281,7 +282,8 @@ bool MemoryMapped::remap(uint64_t offset, size_t mappedBytes)
 
   // Linux
   // new mapping
-  _mappedView = ::mmap64(NULL, mappedBytes, PROT_READ | PROT_WRITE, MAP_SHARED, _file, offset);
+  //[Unk] _mappedView = ::mmap64(NULL, mappedBytes, PROT_READ | PROT_WRITE, MAP_SHARED, _file, offset);
+  _mappedView = ::mmap(NULL, mappedBytes, PROT_READ | PROT_WRITE, MAP_SHARED, _file, offset);
   if (_mappedView == MAP_FAILED)
   {
     _mappedBytes = 0;
diff --git a/hw/xtensa/esp32.c b/hw/xtensa/esp32.c
index f20af03..1e69771 100644
--- a/hw/xtensa/esp32.c
+++ b/hw/xtensa/esp32.c
@@ -68,7 +68,7 @@ esp_err_t ulp_run(uint32_t entry_point)
 #else
 #include <pthread.h>
 #endif
-#include <error.h>
+//#include <error.h>
 #include "hw/i2c/i2c.h"
 #include "esp32_sha.h"
 
unkphone-4:qemu-xtensa-esp32 roderickgroesbeek$ 

Fix for Error3:
===========
unkphone-4:qemu_esp32 roderickgroesbeek$ echo $CFLAGS
-Wno-return-type
unkphone-4:qemu_esp32 roderickgroesbeek$ 
unkphone-4:qemu_esp32 roderickgroesbeek$ ../qemu-xtensa-esp32/configure --disable-werror --extra-cflags="$CFLAGS"  --prefix=`pwd`/root --target-list=xtensa-softmmu,xtensaeb-softmmu

So I can patch/hack it a little bit to have it compile...
But returning to my original question.
But should MacOS platform be supported very easily?

Ebiroll, 2019-02-06:
Hello,
I have not compiled on a Mac platform yet, but I guess it should be possible.
But then i will add your suggestions. We do not need mmap64 as this is used to map the 4MB flash file for read/write.

Ebiroll, 2019-02-06:
I have committed the fixes you suggested.
Please try some more.  Not sure how well qemu will work.
https://wiki.qemu.org/Hosts/Mac

Ebiroll, 2021-08-09:
It compiles fine now. I added an github action to test that it compiles under MacOS.

closed 2021-08-09 by Ebiroll