Ebiroll/qemu_esp32 #12: build fails with `undefined reference to `major'`

morganrallen, 2019-02-24:
```
/usr/bin/ld: qga/commands-posix.o: in function `dev_major_minor':
/home/morgan/devel/qemu-xtensa-esp32/qga/commands-posix.c:656: undefined reference to `major'
/usr/bin/ld: /home/morgan/devel/qemu-xtensa-esp32/qga/commands-posix.c:657: undefined reference to `minor'
collect2: error: ld returned 1 exit status
```

Ebiroll, 2019-02-24:
What operating system and version are you using when compiling?

morganrallen, 2019-02-24:
Ubuntu 19.04

Running with `qemu-system-xtensa -d guest_errors,unimp  -cpu esp32 -M esp32 -m 4M -s`

Ebiroll, 2019-02-24:
You probably already solved this. 
Do a #include <sys/sysmacros.h>

morganrallen, 2019-02-24:
I actually just removed the references to `major`/`minor` :)))

closed 2019-07-05 by Ebiroll

Jiangdiii, 2020-06-30:
I met this Problem,too. My OS is Unbuntu 20.04 LTS. 
After I add #include <sys/sysmacros.h>. I solved this issiue.

TaoTaoNie, 2021-09-25:
> I met this Problem,too. My OS is Unbuntu 20.04 LTS.
> After I add #include <sys/sysmacros.h>. I solved this issiue.

which file to add #include <sys/sysmacros.h>?

DanielWood, 2021-10-08:
> > I met this Problem,too. My OS is Unbuntu 20.04 LTS.
> > After I add #include <sys/sysmacros.h>. I solved this issiue.
> 
> which file to add #include <sys/sysmacros.h>?

commands-posix.c