Ebiroll/qemu_esp32 #13: Cannot send a receive on the same UART

morganrallen, 2019-02-24:
It appears the serial interface cannot send and receive on the same UART.

The behavior can be see in `esp-idf/examples/peripherals/uart/uart_events`

Loading this example on a real ESP32 with a USB bridge connected to `UART_0` anything typed on the terminal will be echo with `uart_write_bytes` and a `ESP_LOGI` call will be made on the default `CONSOLE_UART`, also `UART_0`

Loading this same example in `qemu_esp32` will do nothing. But if `uart_events/main/uart_events_example_main.c` is edited to change...
```
#define EX_UART_NUM UART_NUM_0
```
to
```
#define EX_UART_NUM UART_NUM_1
```

It will work using two instances of `nc` connected to `8880` and `8881`. In this case, anything typed into `8881` will be printed to `8880`

morganrallen, 2019-02-24:
This is on Ubuntu 19.04 running with `qemu-system-xtensa -d guest_errors,unimp  -cpu esp32 -M esp32 -m 4M -s`

Ebiroll, 2019-02-24:
Maybe 
nc 127.0.0.0:8880  |  nc 127.0.0.1:8881 | cat <&0
¢ould have worked as a bridge. I will test this later

Ebiroll, 2019-02-24:
OK. You are right. I will check this later. 
The qemu uart emulation needs some fixing and this is a good example.

morganrallen, 2019-02-24:
OK! Thanks @Ebiroll let me know if there is any other details I can help with.
I have an upcoming project with a large scale text interface that's I prefer not to have to flash
for every minor update. Look forward to digging into qemu-esp32 more. Thanks.

Ebiroll, 2019-02-24:
Good idea. it will probably save you lots of time not having to flash constantly. You should also look at the possibility to run the qemu gdb debugger as it gives you a quick way of finding logic errors and stepping into the esp-idf code.