Ebiroll/qemu_esp32 #18: An Error about the port

coreader, 2019-08-26:
When running the instruction, there is an error. Perhaps, I should 'make menuconfig' to change the port, but how I get the corresponding port? Thanks!
`~/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 -b 921600 -p /dev/ttyUSB0 dump_mem 0x40000000 0x000C2000 rom.bin`
Error：
serial.serialutil.SerialException: [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'

Ebiroll, 2019-08-26:
Hello. You should use the same port (/dev/ttyUSB0) as when you program the esp32.  lsusb -v could give you a hint. lsmod | grep cp210x . Also try sudo ~/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 -b 921600 -p /dev/ttyUSB0 dump_mem 0x40000000 0x000C2000 rom.bin

Ebiroll, 2019-08-26:
I guess, only when dumping the rom files.

coreader, 2019-08-26:
Thanks for your reply. Sorry, I missed the instructions, the program is running on macOS. I don't know how to find the corresponding port. Does the port require physical hardware support?

closed 2019-11-13 by Ebiroll