Ebiroll/qemu_esp32 #19: Debugging with VS Code

21212124, 2019-11-12:
In the debugging part, i do not understand how is it done.
Can you explain me with more details?

Ebiroll, 2019-11-12:
Sorry, but it is a bit outdated. I will update the info.
Please look here for updated info  https://github.com/espressif/esp-idf/issues/303
It works pretty good when you have it up and running. 
The idea is to attach to qemu rather than OpenOCD on port 3333
 "target remote :1234",

21212124, 2019-11-12:
`make -j8 run GDBSERVER=y`
`(qemu) c`
Normal world: 
`gdbserver :12345 optee_example_hello_world`

in other terminal:
`op-tee/out-br/host/bin/aarch64-buildroot-linux-gnu-gdb optee_example_hello_world`
`(gdb) set sysroot op-tee /out-br/host/aarch64-buildroot-linux-gnu/sysroot`
`(gdb) target remote :12345`

but you won't get connect to VS code, any suggestions @Ebiroll ?

Ebiroll, 2019-11-12:
Yes, qemu start a builtin gdbserver. in qemu_world 
(gdb) target remote:1234
It is good start qemu with the options -s -S to start target in a stopped state.

21212124, 2019-11-12:
how to connect with VS code, with F5?

Ebiroll, 2019-11-12:
You add the file .vscode/launch.json
Åress the debug icon Ctr-shift-D

// ======================== launch.json ========================
{
  // Use IntelliSense to learn about possible attributes.
  // Hover to view descriptions of existing attributes.
  // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
  "version": "0.2.0",
  "configurations": [
    {
      "type": "gdb",
      "request": "launch",
      "name": "Debug ESP32",
      "target": "build/${workspaceFolderBasename}.elf",
      "cwd": "${workspaceFolder}",
      "gdbpath": "${env:HOME}/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gdb", // Path of the toolchain
      "autorun": [
        "target remote :1234", // Connect to qemu
        "flushregs", // Force GDB to get new state from the target
        "c" // Resume the program. It will then stop at breakpoint inserted at app_main
      ]
    }
  ]
}

21212124, 2019-11-12:
I do not get it.
What sequence of steps do I have to follow with qemu?

Ebiroll, 2019-11-12:
If you folllow the instructions in the readme you should be able to start as
xtensa-softmmu/qemu-system-xtensa   -cpu esp32 -M esp32  -s  > io.txt
The -s option sets up a gdbserver

Ebiroll, 2019-11-12:
xtensa-softmmu/qemu-system-xtensa -cpu esp32 -M esp32 -s -S > io.txt
This is better as execution halts until debugger has attached

closed 2019-11-13 by 21212124