Ebiroll/qemu_esp32 #20: Build fails with  unknown option --disable-capstone

sebastianene07, 2020-02-18:
Following the README.md instructions on Ubuntu 18.04 will generate an error.

netmonk, 2020-05-30:
Just remove it. 
But make is failing on gtk-3 library not found

Ebiroll, 2020-05-30:
Did you install all these?
sudo apt-get update && sudo apt-get upgrade
sudo  apt-get install build-essential
sudo apt-get install libpixman-1-0 libpixman-1-dev 
apt-get install libglib2.0-dev zlib1g-dev
My Dockerfile builds fine, https://github.com/Ebiroll/qemu_esp32/blob/master/Dockerfile
apt-get install    xz-utils libpixman-1-0 libpng16-16 libjpeg8 libglib2.0

einar-lanfranco, 2020-09-25:
Hi Ebiroll in your Dockerfile, this file doestn exist and build fails,  wget https://github.com/Ebiroll/qemu-xtensa-esp32/suites/317165267/artifacts/360727