Ebiroll/qemu_esp32 #23: Need rom.bin and rom1.bin

markmal, 2020-08-23:
Hi Ebiroll,

Thanks for the project!

in installation manual I see that rom.bin and rom1.bin should be dumped from a physical hardware board.
I do not have that yet. Just ordered, awaiting.
Meantime I wanted to try some examples.

I tried to google them, but links are outdated.
Is it possible you place your rom.bin and rom1.bin to github?

Ebiroll, 2020-08-24:
Hello.
Try espressifs qemu. The have a dump
https://github.com/espressif/qemu
Real Therese wiki as well as my reade
GL

markmal, 2020-08-25:
I've downloaded current release tar. But how to run it? It asks for a flash
`cd quemu`
`bin/qemu-system-xtensa -nographic -machine esp32 -drive 'file=share/qemu/esp32-r0-rom.bin,if=mtd,format=raw'`

Not initializing SPI Flash
qemu-system-xtensa: Error: -bios argument not set, and ROM code binary not found

Ebiroll, 2020-08-26:
You have to copy the rom dump from the source directory to the bin dir.
I do not remember the exact name. Will add that info to the readme
tomorrow.

markmal, 2020-08-26:
Another issue. I tried to compile qemu_flash following Quick Start step 4. But toflash.c is not in qemu_esp32

Ebiroll, 2020-08-26:
The rom location was in the readme. 
cp ../esp-develop/pc-bios/esp32-r0-rom.bin .
The toflash file is located here,
https://github.com/Ebiroll/qemu_esp32/blob/master/toflash.c 
https://github.com/Ebiroll/qemu_esp32/blob/master/toflash-cmake.c

markmal, 2020-08-26:
Thanks I've got toflash. I've built hello_world.bin in Eclipse ESP32-IDF, then flashed to hello_world.bin to esp32flash.bin then started Qemu
./qemu-system-xtensa -d guest_errors,unimp -machine esp32 -bios esp32-r0-rom.bin -drive 'file=esp32flash.bin,if=mtd,format=raw'
it returned 
esp32_init
Socket created
bind done
Waiting for incoming connections...
Waiting for incoming connections...
Socket created
bind done
Waiting for incoming connections...
Waiting for incoming connections...
Socket created
bind done
Waiting for incoming connections...
Waiting for incoming connections...
Open net
trace_sysbus_open_eth_init
esp32flash.bin @ F1E00000
Flash map data to  00000000 to memory, 3F400000
   Can't open 'rom.bin' for reading.
   Can't open 'rom1.bin' for reading.
Warning: Orphaned drive without device: id=mtd0,file=esp32flash.bin,if=mtd,bus=0,unit=0
trace_sysbus_open_eth_reset

Ebiroll, 2020-08-26:
Oh. OK. You are getting closer.
You have built my version of qemu. You should use this version that is the latest version of esp32 with rom provided by espressif.
https://github.com/Ebiroll/qemu-xtensa-esp32s2

Ebiroll, 2020-08-26:
Complete instructions
git clone https://github.com/Ebiroll/qemu-xtensa-esp32s2 
mkdir qemu_espressif
cd qemu_espressif
 ../qemu-xtensa-esp32s2/configure --target-list=xtensa-softmmu     --enable-debug --enable-sanitizers     --disable-strip     --disable-capstone --disable-vnc
make
cp ../esp-develop/pc-bios/esp32-r0-rom.bin .
Make sure you have a usable esp32flash.bin file then start with
xtensa-softmmu/qemu-system-xtensa -nographic -M esp32 -drive file=esp32flash.bin,if=mtd,format=raw -s

markmal, 2020-08-26:
I've successfully built ~/eclipse-esp-workspace/test-hello/hello-world.bin
Then I did following:

cd ~/eclipse-esp-workspace/test-hello/
/fs01/home/mark/opt/esp/qemu_esp32/qemu_flash build/hello-world.bin

4+0 records in
4+0 records out
4194304 bytes (4.2 MB, 4.0 MiB) copied, 0.00894205 s, 469 MB/s
   Can't open 'build/partitions_singleapp.bin' for reading.

cd /fs01/home/mark/opt/
git clone https://github.com/Ebiroll/qemu-xtensa-esp32s2
mkdir qemu_ebirol
cd qemu_ebirol
../qemu-xtensa-esp32s2/configure --target-list=xtensa-softmmu --enable-debug --enable-sanitizers --disable-strip --disable-capstone --disable-vnc
make

cd /fs01/home/mark/opt/qemu_ebirol
cp ~/eclipse-esp-workspace/test-hello/esp32flash.bin ./
xtensa-softmmu/qemu-system-xtensa -nographic -M esp32 -drive file=esp32flash.bin,if=mtd,format=raw -s

Adding SPI flash device
==26809==WARNING: ASan doesn't fully support makecontext/swapcontext functions and may produce false positives in some cases!
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6984
load:0x40078000,len:13072
ho 0 tail 12 room 4
load:0x40080400,len:3896
entry 0x40080688
I (1925) boot: ESP-IDF v4.1 2nd stage bootloader
I (1946) boot: compile time 21:43:43
I (1987) boot: chip revision: 0
I (2034) boot.esp32: SPI Speed      : 40MHz
I (2045) boot.esp32: SPI Mode       : DIO
I (2053) boot.esp32: SPI Flash Size : 2MB
I (2083) boot: Enabling RNG early entropy source...
I (2115) boot: Partition Table:
I (2133) boot: ## Label            Usage          Type ST Offset   Length
I (2216) boot: End of partition table
E (2227) boot: No bootable app partitions in the partition table
ets Jun  8 2016 00:22:57

Ebiroll, 2020-08-26:
The secondary stage bootloader does not find any bootable partitions.
I think It could be that the bootloader is too big or that the partition
table is not flashed correctly. I think it might be that the esptool adds a
checksum to the partition table.
Try setting other bootloader options with menucofig.

Ebiroll, 2020-08-26:
Yes. Note the error massage. Can't open 'build/partitions_singleapp.bin'
for reading.

qemu_flash expects the partitiontable in that location.
It is probably located elsewhere.

markmal, 2020-08-26:
That are all .bins in the project

mark@ustudio:~/eclipse-esp-workspace/test-hello$ find . -name "*bin"
./esp32flash.bin
./build/hello-world.bin
./build/bootloader/CMakeFiles/3.13.4/CMakeDetermineCompilerABI_C.bin
./build/bootloader/CMakeFiles/3.13.4/CMakeDetermineCompilerABI_CXX.bin
./build/bootloader/CMakeFiles/feature_tests.bin
./build/bootloader/bootloader.bin
./build/esp32flash.bin
./build/CMakeFiles/3.13.4/CMakeDetermineCompilerABI_C.bin
./build/CMakeFiles/3.13.4/CMakeDetermineCompilerABI_CXX.bin
./build/CMakeFiles/feature_tests.bin
./build/partition_table/partition-table.bin

I see it has ./build/partition_table/partition-table.bin. Is there a way either to:
a - provide qemu with this bin
b - build the project the way that generates partitions_singleapp.bin
?

Ebiroll, 2020-08-27:
Tje file qemu_esp32/toflash-cmake.c uses this location.
Compile thus inte qemu flash instreaf.

markmal, 2020-08-27:
THANK YOU! 
IT WORKS!

Ebiroll, 2020-09-09:
OK. With the latest sourcecode I have also made it possible to use the rom file provided from espressif.
wget https://github.com/espressif/qemu/raw/esp-develop/pc-bios/esp32-r0-rom.bin
mv esp32-r0-rom.bin rom.bin

closed 2020-09-09 by Ebiroll

redfast00, 2023-06-22:
More permanent link https://github.com/espressif/qemu/raw/0ff3da8d3c797dcf33a45c419204f39f684376cf/pc-bios/esp32-r0-rom.bin