Ebiroll/qemu_esp32 #24: flash read err, 1000

shailes, 2020-09-08:
Thanks for creating such a fantastically useful project.

I'm seeing a problem with the QEMU version here. I've built apps using a number of the example programs, as well as the hello_world example, all compiled against the espressif esp-idf (v4.1). But they won’t run on this fork of QEMU even though they do run on the espressif fork. 

**Creating the app (I’ve tried it with a number with the same result):**
$ pwd
/home/user/esp/qemu_esp32/examples/hello_world
$ make menuconfig
> Set [*] Run FreeRTOS only on first core      
>     [ ] Make exception and panic handlers JTAG/OCD aware 

$ make
$ gcc ../../toflash.c -o qemu_flash 
$ ./qemu_flash build/hello-world.bin

**Ebiroll QEMU**
$ xtensa-softmmu/qemu-system-xtensa -d guest_errors,unimp  -cpu esp32 -M esp32 -m 4M  -s > /dev/null

esp32flash.bin @ 4DE00000
VNC server running on 127.0.0.1:5900
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK

**Espressif QEMU**
I don't see the same behaviour when using the espressif version of QEMU – exactly the same flash loads fine.

Hello world!
This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 0, 2MB external flash
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
Restarting in 4 seconds...
Restarting in 3 seconds...
Restarting in 2 seconds...
Restarting in 1 seconds...
Restarting in 0 seconds...
Restarting now.

Ebiroll, 2020-09-08:
Thanks for the Info. 
I have noticed a similar behavior and it could be related to later versions of the boot ROM. 
Do you know if you dumped a rev3 of the esp32 rom?  

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
Please try without  pipe to  /dev/null and see if there is some more information there...

Ebiroll, 2020-09-08:
Espressif has released the r0 rom.
https://github.com/espressif/qemu/blob/esp-develop/pc-bios/esp32-r0-rom.bin
I will try using that as rom0.bin and rom1.bin.

shailes, 2020-09-08:
> Thanks for the Info.
> I have noticed a similar behavior and it could be related to later versions of the boot ROM.
> Do you know if you dumped a rev3 of the esp32 rom?
>
> rst:0x10 [...]

I used two different (but recent) boards with the same result, but don't know what revision. I'll see if I can dig out something older, Attached the output, which I've truncated in places. It's not obviously useful as far as I can see.

esp32_init
Socket created
bind done
Waiting for incoming connections...
Waiting for incoming connections...
Socket created
bind done
Waiting for incoming connections...
Waiting for incoming connections...
Open net
trace_sysbus_open_eth_init
Socket created
bind done
Waiting for incoming connections...
Waiting for incoming connections...
Flash map data to  00000000 to memory, 3F400000
trace_sysbus_open_eth_reset
io read 44 DPORT_PRO_CACHE_CTRL1_REG  3ff00044=8E6
io write 44,8e6 
 DPORT_PRO_CACHE_CTRL1_REG  3ff00044  8e6
io read 48034 RTC_CNTL_RESET_STATE_REG 3ff48034=3390
io read 48088 RTC_CNTL_DIG_ISO_REG 3ff48088=00000000
io write 48088,0 
RTC_CNTL_DIG_ISO_REG 3ff48088
io read 48088 RTC_CNTL_DIG_ISO_REG 3ff48088=00000000
io write 48088,400 
RTC_CNTL_DIG_ISO_REG 3ff48088
io read 5f06c TIMG_RTCCALICFG1_REG 3ff5f06c=0x25
io read 5a010 EFUSE_BLK0_RDATA4_REG 3ff5a010=01
io read 480b4 RTC_CNTL_STORE5_REG 3ff480b4=00000000
io read 480b4 RTC_CNTL_STORE5_REG 3ff480b4=00000000
io write 480b4,0 
RTC_CNTL_STORE5_REG 3ff480b4 0
io read 48474 
io write 48474,0 
io read 48474 
io write 48474,0 
io read 49088 
io write 49088,0 
io read 49088 
io write 49088,0 
io write 18c,5 
DPORT_PRO_UART_INTR_MAP_REG 5
io read 44038 GPIO_STRAP_REG 3ff44038=0x13
io read 480b0 RTC_XTAL_FREQ_REG 3ff480b0=280028
io read 44038 GPIO_STRAP_REG 3ff44038=0x13
io read 44038 GPIO_STRAP_REG 3ff44038=0x13
io read 44038 GPIO_STRAP_REG 3ff44038=0x13
io read 44038 GPIO_STRAP_REG 3ff44038=0x13
io read 44038 GPIO_STRAP_REG 3ff44038=0x13
io read 44038 GPIO_STRAP_REG 3ff44038=0x13
io read 44038 GPIO_STRAP_REG 3ff44038=0x13
io read 44038 GPIO_STRAP_REG 3ff44038=0x13
io read 44038 GPIO_STRAP_REG 3ff44038=0x13
io read 44038 GPIO_STRAP_REG 3ff44038=0x13
io read 44038 GPIO_STRAP_REG 3ff44038=0x13
io read 44038 GPIO_STRAP_REG 3ff44038=0x13
io read 44038 GPIO_STRAP_REG 3ff44038=0x13
io read 44038 GPIO_STRAP_REG 3ff44038=0x13
io read 44038 GPIO_STRAP_REG 3ff44038=0x13
io read 5f06c TIMG_RTCCALICFG1_REG 3ff5f06c=0x25
io read 5a010 EFUSE_BLK0_RDATA4_REG 3ff5a010=01
[... 200 lines cut ...]
io read 44 DPORT_PRO_CACHE_CTRL1_REG  3ff00044=8E6
io read 5c  DPORT_APP_CACHE_CTRL1_REG  3ff0005C=000008E6
io read 5c  DPORT_APP_CACHE_CTRL1_REG  3ff0005C=000008E6
io write 5c,8ff 
 DPORT_APP_CACHE_CTRL1_REG  3ff0005C= 8ff
io read 44 DPORT_PRO_CACHE_CTRL1_REG  3ff00044=8E6
io write 44,8ff 
 DPORT_PRO_CACHE_CTRL1_REG  3ff00044  8ff
Flash map data to  00000000 to memory, 3F400000
io write 10000,0 
 MMU CACHE  3ff10000  0
io write 44,8e6 
 DPORT_PRO_CACHE_CTRL1_REG  3ff00044  8e6
io write 5c,8e6 
 DPORT_APP_CACHE_CTRL1_REG  3ff0005C= 8e6
io read 44 DPORT_PRO_CACHE_CTRL1_REG  3ff00044=8E6
io write 44,8e6 
 DPORT_PRO_CACHE_CTRL1_REG  3ff00044  8e6
io read 40  DPORT_PRO_CACHE_CTRL_REG  3ff00040=00000020
io write 40,28 
DPORT_PRO_CACHE_CTRL_REG 3ff00040
io read 5a018 EFUSE_BLK0_RDATA6_REG 3ff5a018=01
wifi read 0x1c 
wifi read 0x1c 
... carries on ad infinitum

Ebiroll, 2020-09-08:
Thanks. I think you are using REV3 (or 2) of the rom.
This is new.....
io read 5a018 EFUSE_BLK0_RDATA6_REG 3ff5a018=01
Rev 0 or 1 does not read this information, 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/efuse.html
Thanks for the information, I will update the readme. 

You can probably read espfuse with
espefuse.py -p PORT summary

Ebiroll, 2020-09-08:
To get rev0 rom do, 
wget https://github.com/espressif/qemu/raw/esp-develop/pc-bios/esp32-r0-rom.bin
mv esp32-r0-rom.bin rom.bin

Ebiroll, 2020-09-08:
I also updated the source so that rom1 is no longer required.

shailes, 2020-09-09:
Thanks very much - unfortunately, I still have a problem. I've pulled the new source and the rev0 rom but, on running, I get an indication that the rom is corrupt or truncated:

esp32_init
Socket created
bind done
Waiting for incoming connections...
Waiting for incoming connections...
Socket created
bind done
Waiting for incoming connections...
Waiting for incoming connections...
Open net
trace_sysbus_open_eth_init
Socket created
bind done
Waiting for incoming connections...
Waiting for incoming connections...
esp32flash.bin @ 00600000
Flash map data to  00000000 to memory, 3F400000
 File 'rom.bin' is truncated or corrupt.
VNC server running on 127.0.0.1:5900
...

Ebiroll, 2020-09-09:
Yes, I should probably remove that printout, but then it started for me. Will test more later.

shailes, 2020-09-09:
I'm still getting "flash read err, 1000" - I'll also have a play.

Ebiroll, 2020-09-09:
Ok. Can you send the esp32flash.bin file? Preferably compressed. Thanks

shailes, 2020-09-09:
Thanks - attached. I’ve tried a range of variants in creating this and running it, but I’ve clearly not hit on the right one.

Ebiroll, 2020-09-09:
[esp32flash.bin.gz](https://github.com/Ebiroll/qemu_esp32/files/5194982/esp32flash.bin.gz)
I could not see the file. If you have time. Try starting with this one.

Ebiroll, 2020-09-09:
esp32flash.bin @ 1EC00000
VNC server running on 127.0.0.1:5900
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6440
load:0x40078000,len:11368
load:0x40080400,len:6636
entry 0x40080768
I (3) boot: ESP-IDF v4.1-dev-58-g02c7c3885 2nd stage bootloader
I (3) boot: compile time 12:16:56
I (4) boot: Enabling RNG early entropy source...
I (4) boot: SPI Speed      : 40MHz
I (4) boot: SPI Mode       : DIO
I (4) boot: SPI Flash Size : 2MB
I (4) boot: Partition Table:
I (4) boot: ## Label            Usage          Type ST Offset   Length
I (4) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (4) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (4) boot:  2 factory          factory app      00 00 00010000 00100000
I (4) boot: End of partition table
I (4) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x11e18 ( 73240) map
I (14) esp_image: segment 1: paddr=0x00021e40 vaddr=0x3ffb0000 size=0x02434 (  9268) load
I (16) esp_image: segment 2: paddr=0x0002427c vaddr=0x40080000 size=0x00400 (  1024) load
I (16) esp_image: segment 3: paddr=0x00024684 vaddr=0x40080400 size=0x0b98c ( 47500) load
I (24) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x51594 (333204) map
I (69) esp_image: segment 5: paddr=0x000815b4 vaddr=0x4008bd8c size=0x032cc ( 13004) load
HOST RER TBD
I (74) boot: Loaded app from partition at offset 0x10000
I (74) boot: Disabling RNG early entropy source...
I (74) cpu_start: Pro cpu up.
I (75) cpu_start: Application information:
I (75) cpu_start: Project name:     sigrok
I (75) cpu_start: App version:      1371383-dirty
I (75) cpu_start: Compile time:     Oct  7 2019 16:12:43
I (75) cpu_start: ELF file SHA256:  506e8a964ce9c857...
I (75) cpu_start: ESP-IDF:          v4.1-dev-58-g02c7c3885
I (75) cpu_start: Single core mode
I (75) heap_init: Initializing. RAM available for dynamic allocation:
I (75) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (75) heap_init: At 3FFB8DE8 len 00027218 (156 KiB): DRAM
I (75) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (75) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (75) heap_init: At 4008F058 len 00010FA8 (67 KiB): IRAM
I (75) cpu_start: Pro cpu start user code
esp32_i2c_interruptSet: new IRQ val 0xed377bc0
TBD(pc = 4008c44f): /home/olas/qemu-xtensa-esp32/target-xtensa/translate.c:1409
I (215) spi_flash: detected chip: issi
I (215) spi_flash: flash io: dio
W (215) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (215) cpu_start: Chip Revision: 0
NV_INIT_CALLED

I (215) cpu_start: Starting scheduler on PRO CPU.
I (216) sigrok: Setting UART configuration number 0...
I (216) uart: queue free spaces: 20
Running in qemu
ethoc: num_tx: 8 num_rx: 8
TCP/IP initializing...
TCP/IP initialized.
got ip:192.168.4.3
Applications started.
I (216) sigrok: Setting UART configuration number 1...
I (216) uart: queue free spaces: 10
I (216) gpio: GPIO[12]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (216) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (216) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (216) gpio: GPIO[15]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0

shailes, 2020-09-09:
OK, found the issue. The problem seems to be that qemu doesn't pick up the ~/qemu_esp32 file that gets written with qemu_flash. If instead I manually copy this file to the qemu directory as esp32flash.bin then it seems to work fine. 

Thanks very much for the help in sorting this out.

closed 2020-09-09 by shailes