Ebiroll/qemu_esp32 #30: How to use basic_webserver example

cartu32, 2021-05-16:
Hi, I wanted to run the example number 14_basic_webserver, but I don't know if I'm doing it correctly, because I can't get it to connect to a web browser. So I wanted to ask if I am doing something wrong.
The steps that I follow are the following.

1. The example I am running inside a docker container. Inside this container install and configure the following:

     •	ESP-IDF SDK version 4.2
     •	The Qemu tool from git: //github.com/Ebiroll/qemu-xtensa-esp32
     •	download and configure the git repository: //github.com/Ebiroll/qemu_esp32

2.  Then I went to the example directory, and did the following
     •	I copied the files rom.bin and rom1.bin in the directory of the example
     •	Then I did menuconfig
     •	make
     •	Then execute gcc toflash.c -o qemu_flash
     •	After ./qemu_flash build / 14_basic_webserver.elf

3. Once the esp32flash.bin binary is generated, I run qemu as follows.

`qemu-system-xtensa -d guest_errors,unimp -cpu esp32 -M esp32 -m 4M -net nic,model=vlan0 -net user,id=simnet,ipver4=on,net=172.17.0.0/24,host=172.17.0.2,hostfwd=tcp::10800-172.17.0.2:80 -net dump,file=/tmp/vm0.pcap -s  > io.txt`

So the Docker's console shows me the following:

(92) esp_image: segment 5: paddr=0x000b7104 vaddr=0x4008b2b0 size=0x08dfc ( 36348) load
HOST RER TBD
I (101) boot: Loaded app from partition at offset 0x10000
I (101) boot: Disabling RNG early entropy source...
I (101) cpu_start: Pro cpu up.
I (102) cpu_start: Application information:
I (102) cpu_start: Project name:     14_basic_webserver
I (102) cpu_start: App version:      2aaea3b-dirty
I (102) cpu_start: Compile time:     May 14 2021 22:50:32
I (102) cpu_start: ELF file SHA256:  bbb22de069bde0d7...
I (102) cpu_start: ESP-IDF:          v3.3.2-dirty
I (102) cpu_start: Starting app cpu, entry point is 0x4008128c
I (0) cpu_start: App cpu up.
I (1782) heap_init: Initializing. RAM available for dynamic allocation:
I (1782) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1783) heap_init: At 3FFBA910 len 000256F0 (149 KiB): DRAM
I (1783) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1783) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1784) heap_init: At 400940AC len 0000BF54 (47 KiB): IRAM
I (1784) cpu_start: Pro cpu start user code
esp32_i2c_interruptSet: new IRQ val 0x601e82b0
TBD(pc = 400940a7): /root/qemu-xtensa-esp32/target-xtensa/translate.c:1409
I (590) cpu_start: Starting scheduler on PRO CPU.
I (990) cpu_start: Starting scheduler on APP CPU.
NV_INIT_CALLED

4.  Then I open another console within Docker and try to connect to the web server with netcat, as follows

    `nc 172.17.0.2 10800`

But what I see is that it does not connect to the server, since nothing is shown. On the other hand, the qemu console always shows the same thing. Does not advance from NV_INIT_CALLED. I also tried to connect using Putty from outside the container, but the same thing happens to me.

For that reasonn I wanted to ask if I am doing something wrong.

From already thank you very much.

Ebiroll, 2021-05-16:
Hello, You seem to bedoing it correctly, but try compiling for only one core, 

When configuring, choose
  Component config  --->   
    FreeRTOS  --->
      [*] Run FreeRTOS only on first core

cartu32, 2021-05-16:
Thanks for the quick reply. I Enable the compilation option to run Freertos only on the first core and i can advance the execution of the example. But I still can't connect from netcat or putty. The qemu console shows me the following.

I (99) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (99) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (99) heap_init: At 4009355C len 0000CAA4 (50 KiB): IRAM
I (99) cpu_start: Pro cpu start user code
esp32_i2c_interruptSet: new IRQ val 0x8dd832b0
TBD(pc = 40093557): /root/qemu-xtensa-esp32/target-xtensa/translate.c:1409
I (263) cpu_start: Starting scheduler on PRO CPU.
NV_INIT_CALLED

nvs_flash_init
Running in qemu
Running in qemu
ethoc: num_tx: 8 num_rx: 8
TCP/IP initializing...
TCP/IP initialized.
Applications started.
Waiting for connection to the wifi network...
 Connected

IP Address:  0.0.0.0
Subnet mask: 0.0.0.0
Gateway:     0.0.0.0
Running in qemu
HTTP Server listening...

Seems not detecting IP of docker container. So I wanted to ask what the problem could be.

From already thank you very much.