Ebiroll/qemu_esp32 #32: MacOS simple compilation error "error: conflicting types for 'esp32_i2c_interruptSet'"

rompinpl, 2021-08-07:
Hi,
during compilation I have got :
.. /qemu-xtensa-esp32/hw/xtensa/esp32.c:2835:13: error: conflicting types for 'esp32_i2c_interruptSet'
extern void esp32_i2c_interruptSet(qemu_irq new_irq);

.. /qemu-xtensa-esp32/hw/xtensa/esp32.c:2476:12: note: previous implicit declaration is here
                   esp32_i2c_interruptSet(PROcpu->env.irq_inputs[val]);

solution: move at beginning  lines with "extern ..."

Ebiroll, 2021-08-09:
Thanks, I fixed the problem and added a test Mac compile action to test that it does compile.

closed 2021-08-09 by Ebiroll

rompinpl, 2021-08-09:
Thx ;)
And thx for project :)

Ebiroll, 2021-08-09:
If you want to dive deeper learn about ghidra. https://olof-astrand.medium.com/enter-home-dragon-with-ghidra-3ed7ddf75935
GL