Ebiroll/qemu_esp32 #33: `make` command failes

Miraj98, 2021-08-20:
According the README I was able to execute these commands successfully
```
git clone git://github.com/Ebiroll/qemu-xtensa-esp32
mkdir qemu_esp32
cd qemu_esp32
../qemu-xtensa-esp32/configure --disable-werror --prefix=`pwd`/root --target-list=xtensa-softmmu,xtensaeb-softmmu
```

But on trying to run `make` it fails the following error trace
```
22 warnings generated.
  CXX   xtensa-softmmu/hw/xtensa/MemoryMapped.o
In file included from /Users/miraj/qemu-xtensa-esp32/hw/xtensa/MemoryMapped.cpp:7:
In file included from /Users/miraj/qemu-xtensa-esp32/hw/xtensa/MemoryMapped.h:20:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/string:506:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/string_view:175:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/__string:57:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/algorithm:639:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/initializer_list:46:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/cstddef:37:
/Users/miraj/qemu-xtensa-esp32/version:1:1: error: expected unqualified-id
2.7.0
^
In file included from /Users/miraj/qemu-xtensa-esp32/hw/xtensa/MemoryMapped.cpp:7:
In file included from /Users/miraj/qemu-xtensa-esp32/hw/xtensa/MemoryMapped.h:20:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/string:506:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/string_view:175:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/__string:57:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/algorithm:639:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/initializer_list:46:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/cstddef:49:9: error: no member named 'ptrdiff_t' in the global namespace
using ::ptrdiff_t;
      ~~^
In file included from /Users/miraj/qemu-xtensa-esp32/hw/xtensa/MemoryMapped.cpp:7:
In file included from /Users/miraj/qemu-xtensa-esp32/hw/xtensa/MemoryMapped.h:20:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/string:506:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/string_view:175:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/__string:57:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/algorithm:640:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/type_traits:418:
/Users/miraj/qemu-xtensa-esp32/version:1:1: error: expected unqualified-id
2.7.0
^
In file included from /Users/miraj/qemu-xtensa-esp32/hw/xtensa/MemoryMapped.cpp:7:
In file included from /Users/miraj/qemu-xtensa-esp32/hw/xtensa/MemoryMapped.h:20:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/string:506:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/string_view:175:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/__string:57:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/algorithm:642:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/utility:205:
/Users/miraj/qemu-xtensa-esp32/version:1:1: error: expected unqualified-id
2.7.0
^
In file included from /Users/miraj/qemu-xtensa-esp32/hw/xtensa/MemoryMapped.cpp:7:
In file included from /Users/miraj/qemu-xtensa-esp32/hw/xtensa/MemoryMapped.h:20:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/string:506:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/string_view:175:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/__string:57:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/algorithm:643:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/memory:668:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/typeinfo:60:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/exception:83:
/Users/miraj/qemu-xtensa-esp32/version:1:1: error: expected unqualified-id
2.7.0
^
In file included from /Users/miraj/qemu-xtensa-esp32/hw/xtensa/MemoryMapped.cpp:7:
In file included from /Users/miraj/qemu-xtensa-esp32/hw/xtensa/MemoryMapped.h:20:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/string:506:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/string_view:175:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/__string:57:
[... 86 lines cut ...]
In file included from /Users/miraj/qemu-xtensa-esp32/hw/xtensa/MemoryMapped.h:20:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/string:506:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/string_view:175:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/__string:57:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/algorithm:643:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/memory:681:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/atomic:1052:73: error: unknown type name 'ptrdiff_t'
_Tp* __cxx_atomic_fetch_add(__cxx_atomic_base_impl<_Tp*> volatile* __a, ptrdiff_t __delta, memory_order __order) _NOEXCEPT {
                                                                        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/atomic:1057:65: error: unknown type name 'ptrdiff_t'
_Tp* __cxx_atomic_fetch_add(__cxx_atomic_base_impl<_Tp*> * __a, ptrdiff_t __delta, memory_order __order) _NOEXCEPT {
                                                                ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/atomic:1073:73: error: unknown type name 'ptrdiff_t'
_Tp* __cxx_atomic_fetch_sub(__cxx_atomic_base_impl<_Tp*> volatile* __a, ptrdiff_t __delta, memory_order __order) _NOEXCEPT {
[... cut ...]
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/v1/atomic:1078:65: error: unknown type name 'ptrdiff_t'
_Tp* __cxx_atomic_fetch_sub(__cxx_atomic_base_impl<_Tp*> * __a, ptrdiff_t __delta, memory_order __order) _NOEXCEPT {
                                                                ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[1]: *** [hw/xtensa/MemoryMapped.o] Error 1
make: *** [subdir-xtensa-softmmu] Error 2
```

Not sure what's the issue. Any help is appreciated

Ebiroll, 2021-08-20:
Try to add #include <stddef.h> in the file that has problem compiling

Miraj98, 2021-08-20:
Not sure what file that would be. I added `#include <stddef.h>` in `qemu-xtensa-esp32/hw/xtensa/MemoryMapped.cpp` but still didn't seem to work

jpitkanen18, 2021-09-03:
You need to install Device Tree Compiler (DTC)
on macOS with homebrew installed run `brew install dct`

hughlv, 2021-09-04:
I am facing the exactly  same issues when trying 'make' on macOS Big Sur. 

Basically following errors:

1. error: expected unqualified-id  -- from MappedMemory.cpp
2. /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/iterator:562:13: error: unknown type name 'ptrdiff_t'
3. /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/v1/iterator:989:54: error: template parameter missing a default argument
          class _Traits = char_traits<_CharT>, class _Distance = ptrdiff_t>

All looked like lib compatibility issues. I've followed the suggestion from @Ebiroll and added 'include <stddef.h>' as the first line of MappedMemory.cpp but got no luck. Still the same errors.

Any clues to solve these issues? Thanks.

Miraj98, 2021-09-04:
> You need to install Device Tree Compiler (DTC)
> on macOS with homebrew installed run `brew install dct`

@jpitkanen18 Tried this but did not work. Getting the same error. Also there is a typo. If I am not wrong it should be `dtc` and not `dct`

fanzhiri, 2021-10-10:
I encountered this error and I solved it .
rename the file : qemu-xtensa-esp32/version
for you : mv /Users/miraj/qemu-xtensa-esp32/version /Users/miraj/qemu-xtensa-esp32/py_version

jpitkanen18, 2021-11-12:
> > You need to install Device Tree Compiler (DTC)
> > on macOS with homebrew installed run `brew install dct`
> 
> @jpitkanen18 Tried this but did not work. Getting the same error. Also there is a typo. If I am not wrong it should be `dtc` and not `dct`

I installed the library but ALSO renamed the version file to version.txt and forgot about it. My bad haha